sinara-hw/HVSUP_ISOL #3: schematic review

filipswit, 2022-04-10:
**HV Module**
- [ ] Shorted +Vin to -Vin
[image]

- [ ] It is recommended to place 1uF **and 100nF** in parallel beside isolation links
- [ ] P3V3 shorted with P3V3p through IC27 and IC28 
[image]

- [ ] Missing decap at IC27
- [ ] MAX1329: `CAVDD and CDVDD filter capacitor can be a low-ESR tantalum or may need to be a combination of a small ceramic and a larger tantalum capacitor
` It would be probabbly good to change one 10uf ceramic to tantalum
**HVPSU ISOL**

- [ ] Useless test point for P3V3A

tprzywoz, 2022-04-27:
- [ ] MAX1329 RSTs pins are push-pull when configured as interrupt outputs so we need to separate them.

filipswit, 2022-05-01:
Thanks @tprzywoz, I've added 2 mos keys.

closed 2022-05-20 by filipswit